mKenfenheuer/steam-deck-windows-usermode-driver #1: Swicd configurator - blank

Gwizz007, 2022-05-29:
I am able to install the driver and it is detected without steam but when I open the swicd configurator I get a blank form with no options to configure. 
Ran in administrative mode, still nothing. 

Shows up in reWASD as a Xbox controller with no back pedals but when the back buttons are pressed it responds.

Can the back buttons be remapped from what you have as default?

Gwizz007, 2022-05-29:
Awesome work btw, thanks.

Gwizz007, 2022-05-29:
Got the config file open in notepad but after adding 
# whitelist=ModernWarefare.exe
# mode=whitelist
I am unable to save the text document. Says I should contact the contact author when I try to replace and save changes it says I do not have permission to open this file.

mKenfenheuer, 2022-05-29:
Let me answer your questions one by one:

1. The configurator is a WIP, so therefore its empty and without any function right now.
2. A remapping of the buttons/axis is on my to do list, but not my top priority right now,
3. Try to save the file to your desktop, then copy it to the directory. (Or open notepad.exe as admin, then open the file using File> Open ... ) The problem here is that UAC is preventing you from saving files there without an elevated prompt (The one where you need to say i want this program to make changes to my system)

Let me know if this works for you.
4.

Gwizz007, 2022-05-29:
Thanks for your quick response. Copying works. Excellent first attempt btw. Will keep following as I am excited to see what you do next with this thing. 
Closed

closed 2022-05-29 by Gwizz007