mKenfenheuer/steam-deck-windows-usermode-driver #2: Buttons / Axis Remapping

mKenfenheuer, 2022-05-29:


Gwizz007, 2022-05-29:
Button remapping is missing for me.
Rumble.
Gyro toggle if possible
If I could get the ... button assignable that would be great.
Love that steam button brings up Xbox quick menu.

mKenfenheuer, 2022-05-30:
@Gwizz007  have a look at release v0.1.2.
It features button/axis mapping and haptic feedback as well as an option to enable/disable mouse emulation per game

Gwizz007, 2022-05-30:
That's some quick work my guy. I will update in a few hours and give you some more feedback. Thanks.

mKenfenheuer, 2022-05-30:
There is also a lot of changes in the config file, have a look at the config wiki entry [here](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/wiki/Configuration-via-config-file).

Let me know if there is anything unclear / missing.

Gwizz007, 2022-05-30:
Awesome work, I went through it and that's some detailed work. 
My one issue is how to activate the stick without touch required if I use thumb grips. 
I tried LeftStickY=LeftStickY

mKenfenheuer, 2022-05-30:
That should work. You can also try  with „None“ as the button name.

Holen Sie sich Outlook für iOS<https://aka.ms/o0ukef>

Gwizz007, 2022-05-31:
Unfortunately, even on the default config file the sticks don't work with touch. There might be a conflict somewhere.

Gwizz007, 2022-06-01:
Any luck with getting the joy stick to function again?

Gwizz007, 2022-06-01:
Seems the sticks are just not tied into the commands, let me know what else I should try

Gwizz007, 2022-06-01:
Oh, I am using the default config file with no changes. 
Also can I get the command for assigning Alt+R to the quick access menu button? Want to use it as the Amd Adrenalin overlay button.

Gwizz007, 2022-06-01:
Got this to work, thanks for the hardwork and the help. Looking for a gui in the future.
Also placed the shortcut to the swicd_driver in my startup folder in hopes of steam button and quick access button use without manually running the driver. Thanks again!!!!
******************
# the general section defines layout independent behaviour such as
# when to emulate gamepad inputs or how to determine if emulation should
# be performed.
[general]
# use blacklist to prevent SWICD from emulating input if a process with the same executable name is running.
# mode must be set to blacklist, if not this setting is ignored
#
#  blacklist=steam.exe

# use whitelist to enable SWICD only if a process with the same executable name is running. 
# mode must be set to whitelist, if not this setting is ignored
#
#  whitelist=ModernWarfare.exe
#  whitelist=Modern Warfare Launcher.exe
#  whitelist=Playnite.exe

# use mode to specify the mode of operation. 
# whitelist will only enable SWICD emulation if a whitelisted process is running.
# blacklist will prevent SWICD emulation if a blacklisted process is running.
#
# DEFAULT: blacklist
#
#  mode=blacklist

# the actions section is used to assign button presses to specific actions.
# actions execute regardless of any emulation in progress.
[actions]
OpenWindowsGameBar=BtnSteam

# use the section "buttons" to customize your global button mappings.
# button mapping is defined as follows:
# 
#  [HardwareButton]=[EmulatedButton]
#
[buttons]
BtnX=BtnX
BtnY=BtnY
BtnA=BtnA
BtnB=BtnB
BtnMenu=BtnStart
BtnOptions=BtnBack
BtnSteam=BtnStart
BtnQuickAccess=None
BtnDpadUp=BtnDpadUp
BtnDpadLeft=BtnDpadLeft
BtnDpadRight=BtnDpadRight
BtnDpadDown=BtnDpadDown
BtnL1=BtnLB
BtnR1=BtnRB
BtnL2=None
BtnR2=None
BtnL4=BtnLS
BtnR4=BtnB
BtnL5=BtnA
BtnR5=BtnA
BtnRPadPress=BtnRS
BtnLPadPress=BtnLS
BtnRPadTouch=None
BtnLPadTouch=None
BtnRStickPress=BtnRS
BtnLStickPress=BtnLS
BtnRStickTouch=None
BtnLStickTouch=None

# use the section "profile" to tweak controller behaviour depending on the executable.
[profile]
DisableLizardMode=false

# use the section "profile" to tweak controller behaviour depending on the executable.
[profile,speed.exe]
DisableLizardMode=true

# use the section "buttons" in combination with an executable name to customize your button mappings per executable.
# button mapping is defined as follows:
# 
#  [HardwareButton]=[EmulatedButton]
#
[buttons,speed.exe]
BtnX=BtnX
BtnY=BtnY
BtnA=BtnA
BtnB=BtnB
BtnMenu=BtnStart
BtnOptions=BtnBack
BtnSteam=BtnStart
BtnQuickAccess=None
BtnDpadUp=BtnDpadUp
BtnDpadLeft=BtnDpadLeft
BtnDpadRight=BtnDpadRight
BtnDpadDown=BtnDpadDown
BtnL1=BtnLB
BtnR1=BtnRB
BtnL2=None
BtnR2=None
BtnL4=BtnA
BtnR4=BtnB
BtnL5=BtnX
BtnR5=BtnY
BtnRPadPress=BtnRS
BtnLPadPress=BtnLS
BtnRPadTouch=None
BtnLPadTouch=None
BtnRStickPress=BtnRS
BtnLStickPress=BtnLS
BtnRStickTouch=None
BtnLStickTouch=None

# use the section "axis" to customize your global button mappings.
# emulation of an axis can be restricted by a button activation 
# e.g. only apply hardware axis RightStickX to emulated axis RightStickX if right stick is touched:
#
#  RightStickX=RightStickX,activate=BtnRStickTouch
#
# axis mapping is defined as follows:
# 
#  [HardwareAxis]=[EmulatedAxis][,activate=ButtonActivator][,inverted=true/false]
#
[axes]
LeftStickX=LeftStickX
LeftStickY=LeftStickY
RightStickX=RightStickX
RightStickY=RightStickY
LeftPadX=LeftStickX,activate=BtnLPadTouch
LeftPadY=LeftStickY,activate=BtnLPadTouch
RightPadX=RightStickX,activate=BtnRPadTouch
RightPadY=RightStickY,activate=BtnRPadTouch
RightPadPressure=None
LeftPadPressure=None
L2=LT
R2=RT
GyroAccelX=None
GyroAccelY=None
GyroAccelZ=None
GyroRoll=None
GyroPitch=None
GyroYaw=None

#use the section "axis" in combination with an executable name to customize your button mappings per executable.
# emulation of an axis can be restricted by a button activation 
# e.g. only apply hardware axis RightStickX to emulated axis RightStickX if right stick is touched:
#
#  RightStickX=RightStickX,activate=BtnRStickTouch
#
# axis mapping is defined as follows:
# 
#  [HardwareAxis]=[EmulatedAxis][,activate=ButtonActivator][,inverted=true/false]
#
[axes,speed.exe]
LeftStickX=LeftStickX
LeftStickY=LeftStickY
RightStickX=RightStickX
RightStickY=RightStickY
LeftPadX=LeftStickX,activate=BtnLPadTouch
LeftPadY=LeftStickY,activate=BtnLPadTouch
RightPadX=RightStickX,activate=BtnRPadTouch
RightPadY=RightStickY,activate=BtnRPadTouch
RightPadPressure=None
LeftPadPressure=None
L2=LT
R2=RT
GyroAccelX=None
GyroAccelY=None
GyroAccelZ=None
GyroRoll=None
GyroPitch=None
GyroYaw=None

mKenfenheuer, 2022-06-06:
Glad to hear it's working now.

> Also can I get the command for assigning CTRL+SHIFT+O to the quick access menu button? Want to use it as the toggle for Amd Adrenalin overlay button.

I'll be working on additional button overlays once i've ported most of the code to c# to have a more stable service with gui etc..

mKenfenheuer, 2022-06-08:
The configurator is now mostly functional. Please try with the latest release.

https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/releases/tag/v0.2-dotnet

Gwizz007, 2022-06-09:
Hey I noticed that the B button while mapped to B also doubles as the start or back button instead of being independent B. Number of users with similar feedback so I am mentioning it. Also happens to me.

mKenfenheuer, 2022-06-09:
Hi, 

is the lizard mode enabled in your default profile?
which release are you on? I published another release yesterday which fixes some button issues.

Gwizz007, 2022-06-09:
Hi,
Thanks for replying so quickly. I am on the latest build with the button fix and lizard mode is on.

Gwizz007, 2022-06-09:
Also assigning B to the back button works as B should but pressing actual B brings up the pause or back menu.

AlexN85, 2022-06-10:
> Got this to work, thanks for the hardwork and the help. Looking for a gui in the future. Also placed the shortcut to the swicd_driver in my startup folder in hopes of steam button and quick access button use without manually running the driver. Thanks again!!! [app_config.conf.txt](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/8863997/app_config.conf.txt) !

How did u get that command to work by the way?

Gwizz007, 2022-06-10:
Please use the latest build, it's way more stable and sits nicely in the system tray. There is also a setting to run at startup in the latest build.

AlexN85, 2022-06-10:
> Please use the latest build, it's way more stable and sits nicely in the system tray. There is also a setting to run at startup in the latest build.

Sorry I meant how did u get the quick access button to work with your amd adrenalin when u press it?

Which release should I use?

Sorry I'm very new to this :(

I want to get that when i press the quick access button it enables say the got back to desktop or triggers the start menu while say in a game? Hope that makes sense

mKenfenheuer, 2022-06-10:
> > Please use the latest build, it's way more stable and sits nicely in the system tray. There is also a setting to run at startup in the latest build.
> 
> Sorry I meant how did u get the quick access button to work with your amd adrenalin when u press it?
> 
> Which release should I use?
> 
> Sorry I'm very new to this :(
> 
> I want to get that when i press the quick access button it enables say the go back to desktop or triggers the start menu while say in a game? Hope that makes sense. I want something like that to trigger so I can uses windows while using a game :)

please have a look at your issue and my reply in #16

Gwizz007, 2022-06-10:
Oh QAM while programmable it's not open to custom shortcuts as yet. I hope he gives us a way to assign keyboard shortcuts to that button. Think adrenalin overlay toggle is crtl shift O. Right now you can only assign it to what's on the list. 

I am using the latest version listed.

AlexN85, 2022-06-10:
> Oh QAM while programmable it's not open to custom shortcuts as yet. I hope he gives us a way to assign keyboard shortcuts to that button. Which he linked above. I think adrenalin overlay toggle is crtl shift O. Right now you can only assign it to what's on the list.
> 
> I am using the latest version listed.

Ahhh ok so u haven't got the adrenalin over to work with that button? Just hope it gets implemented later correct?

Gwizz007, 2022-06-10:
No, it would require him to bind the shortcut similar to how he did windows gamebar. Pretty neat. Like having a Xbox portable. He does good work and pretty fast with the updates and he take suggestions and replies to queries so be patient, he will get to it.

AlexN85, 2022-06-10:
> No, it would require him to bind the shortcut similar to how he did windows gamebar. Pretty neat. Like having a Xbox portable. He does good work and pretty fast with the updates and he take suggestions and replies to queries so be patient, he will get to it.

Ok :) thanks for the reply. I can be patient for greatness work :)

mKenfenheuer, 2022-06-10:
I am thinking about not defining any shortcuts at all.
Instead i can give you the ability to define button actions as you wish.

Like if STEAM and QUICK ACCESS buttons are pressed simultaneously, emulate ALT+F4.
(Define a number of hardware buttons which will trigger a set of keyboard buttons pressed)

This way you can emulate WIN+G (Gamebar) or ALT+R (Radeon settings) or whatever you like

But i need to come up with some nice gui for the configuration. What do you think?

Gwizz007, 2022-06-10:
Sounds perfect, thanks

AlexN85, 2022-06-10:
Amazing thanks sooo much :)

mKenfenheuer, 2022-06-17:
Closing this issue in favor of #21 since the keyboard actions do not have anything to do with this issue anymore and the remapping and gui editor have been implemented.

closed 2022-06-17 by mKenfenheuer