mKenfenheuer/steam-deck-windows-usermode-driver #5: Motion control support for Windows Steam Deck

CasperH2O, 2022-06-02:
Hey there! I'm one of the developers of Handheld Companion/Controller Service, someone came up on our Discord sub channel to ask if your app could be combined with ours to make it possible to use motion control on steam deck through Windows. 

Our application currently supports embedded and external motion sensors, external controllers and some other bits. I saw your emulated device does have some gyro and accelerometer in the interface. Perhaps we can talk, combine some effort maybe?

https://github.com/Valkirie/ControllerService

PS Love your resume website design! I'll be sure to open it next time someone from IT walks past me at work, haha.

mKenfenheuer, 2022-06-02:
Hi, sure! Would you be up for a chat on discord?

Thanks, it indeed has lead to some funny stories and confision!

Holen Sie sich Outlook für iOS<https://aka.ms/o0ukef>

https://github.com/Valkirie/ControllerService

CasperH2O, 2022-06-02:
Yes, we are on Discord: https://discord.gg/6UQ4qMMG

[image]

closed 2022-06-06 by mKenfenheuer