mKenfenheuer/steam-deck-windows-usermode-driver #6: Remapping won't work, or even stay?

Dre2timez, 2022-06-03:
so to start, I want to fully disable the touch pad features it is defaulted by. so when I edit the app_config.conf and set everything involving the pad to [None] it doesn't work and it also gets reverted back to the default settings when I open SWICD_Configurator.exe. Also every .exe in your folder is set as admin if it matters.

I love the drivers itself btw Keep up the great work!

mKenfenheuer, 2022-06-03:
Can you post your config file please?
(Attach it if possible please.)

are you using an executable specific config?

Dre2timez, 2022-06-03:
[app_config.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/8836099/app_config.zip)
I was just using the base config that was created.                                                                                                         I also did try pointing my custom settings to a whitelisted program, Microsoft edge, and nothing changed whilst also overwriting my settings.

closed 2022-06-08 by mKenfenheuer