mKenfenheuer/steam-deck-windows-usermode-driver #8: Issues in Fortnite with default configuration

lwbt, 2022-06-05:
Hello, thank you very much for publishing this project!

I noticed that you already fixed a mistake in the [default configuration](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/commit/573010cbf54e8ff2aad28e730556a8f2d157a7b7). With that I have no more problems in another game, but Fortnite is acting very weird either registering double button presses or trying to perform more than one action when for example trying to press the B button. The menu navigation is also broken to the point where I have to use the touchscreen to leave the game or enter the settings menu. I installed Fortnite through the Epic Games Launcher. I have no problems navigating the menu on Nintendo Switch or PS5, just to rule out major human error. :-) 
 
Also what is lizard mode?

mKenfenheuer, 2022-06-06:
> Fortnite is acting very weird either registering double button presses or trying to perform more than one action when for example trying to press the B button.

The issues you have described are related to the lizard mode (mouse and keyboard emulation).
This means that the game will receive the controller input as well as the mouse and keyboard inputs. Therefore it registeres double inputs. Same will apply if there is steam running and steam input active.

See here for more information:

[https://www.reddit.com/r/SteamController/wiki/index#wiki_steam_not_running_.28aka_lizard_mode.29](https://www.reddit.com/r/SteamController/wiki/index#wiki_steam_not_running_.28aka_lizard_mode.29)

[https://i.imgur.com/P8Q4WHD.png](https://i.imgur.com/P8Q4WHD.png)

Try to disable it using a profile specific for the fortnite executable:

```conf
[profile,fortnite.exe]
DisableLizardMode=true
```

Also make sure steam is not running, it messes with the driver in very nasty ways.

lwbt, 2022-06-06:
I don't have Steam installed on Windows for my Steam Deck.

I tried:

```
[profile,FortniteClient-Win64-Shipping.exe]
DisableLizardMode=true
```

as well es disabling lizard mode for the default profile, but the problem remains. I checked for key assignments and removed some of the keys from the default keyboard mapping in fortnite which might be related, just to be sure. But no luck either.

It is a fresh clean instalaltion of Windows 11 as Windows to Go booted from an external drive with the required Steam Deck drivers. Epic Games Launcher and two games installed, nothing else.

lwbt, 2022-06-06:
I think we can close this as it does not seem to be caused by emulation.

Fortnite itself detects all buttons and axes of the Steam Deck. The A and B buttons do not behave correctly and some others may be too. One just needs to build up muscle memory use L4, R4, L5 and R5 instead. Which is at least for me easier than getting used to the button mapping on the Switch.

I searched a bit, but I guess there is no way around that Epic or the Fortnite developers themselves have to improve how the controller behaves.

Summary: similar like with Steam running in the background, except that you can't turn it off as the game does it all by itself.

lwbt, 2022-06-07:
What I have configured now:

```
[profile,FortniteClient-Win64-Shipping.exe]
DisableLizardMode=true

[buttons,FortniteClient-Win64-Shipping.exe]
BtnMenu=None
BtnOptions=None
BtnSteam=None
```

I took the remaining button mappings from the speed.exe profile. This resolves most of the misbehaving buttons. I can now access sidebar menu (or whatever they call it) in game through the Menu button, which seems to be correct. I'm not sure what the Options button is and what the Quick Access button is. All of them should be mapped to none, but here is what they actually do: The Menu button (hamburger/three bars) opens the in-game menu, like I said. The button which Microsoft calls the View button (two overlapping windows), opens the inventory. Which is not necessary since it is mapped by default to D-pad up, which works fine. So it is mapped twice for some reason. The Steam button and the three dots button open Windows Game Bar. Touchscreen works as a last resort.

mKenfenheuer, 2022-06-08:
Weird, Please try with the latest release.

https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/releases/tag/v0.2-dotnet

closed 2022-06-10 by mKenfenheuer

lwbt, 2022-06-13:
Okay it works with the latest release (v.0.2.3) now.

Note to myself, and for others:

1. Create a new profile from the GUI.
    Executable path, for your convenience, is `"C:\Program Files\Epic Games\Fortnite\FortniteGame\Binaries\Win64\FortniteClient-Win64-Shipping.exe"`
2. Disable Lizard Mode.
    This will cause the controller, e.g. touchpad (mouse cursor) to become unresponsive while Fortnite is starting up. You can still use the touchscreen though. Maybe configuring the touchpad to be ignored would be a better choise. 
3. Swap `BtnStart` and `BtnBack` (the overlapping windows e.g. view button and the three bars button) so that that button functions match the button faces.
    Optionally map `BtnBack` also to the three dots button. (I'm used to press the PS5 touchpad with the right thumb to open the map, that's why and this comes close enough.)

Difference to defaults: 
```
[profile,FortniteClient-Win64-Shipping.exe]
DisableLizardMode=true

[buttons,FortniteClient-Win64-Shipping.exe]
BtnMenu=BtnBack
BtnOptions=BtnStart
BtnQuickAccess=BtnBack
```