mKenfenheuer/steam-deck-windows-usermode-driver #9: Help with build issues?

ReticentRobot, 2022-06-07:
Is this the right place to ask for help with build issues?  I'm not sure what I'm supposed to do to fix this, I think I'm running into the issues under the Build section in the instructions, but not sure what actions I need to take to correct them.

[image]

mKenfenheuer, 2022-06-07:
This is because you do not have the correct build tools version installed. You can install it using the visual studio installer. Also make sure the build output path is set. To do that right click the vscx project, select properties and change the output path there. I cant provide detailed instructions right now, but you might try googling the issues again, those are pretty common ones i would say

But if i were you, i would not waste much time with the c++ version. Have a look at the dotnet-rewrite branch. Its mostly based on c# and some c++ bindings and is more solid and has a nice gui upcoming.

ReticentRobot, 2022-06-07:
Nice! That is this project, correct? https://github.com/mKenfenheuer/sc-controller

Thanks so much for working on this!!

mKenfenheuer, 2022-06-07:
No, this branch here 

https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/tree/dotnet-rewrite

closed 2022-06-07 by ReticentRobot