mKenfenheuer/steam-deck-windows-usermode-driver #15: GUI should show the version number and icons for buttons

lwbt, 2022-06-09:
The GUI looks great!

Putting the version number in the log or somewhere else would help with bug reports I guess. I checked the version number of the executable (1.0) and under installed software (1.3 or 0.1.3, can't exactly remember) both were different to the version number I downloaded from the releases page. Putting the version number in the filename of the download would also be helpful I guess.

I'm still struggling to differentiate some buttons. A, B, X, RT are mostly self explanatory, but for others icons in the button mapping would help so you know what the button with the overlapping windows is. Looks like Microsoft has a font for that (https://support.xbox.com/en-US/help/hardware-network/controller/xbox-one-wireless-controller), not sure if you can use that or Font Awesome or something else appropriate.

Dre2timez, 2022-06-09:
Hey, does your config changed inputs work or do they also revert back after reopening the program? I opened a now closed issue report but the prob still remains.

mKenfenheuer, 2022-06-10:
> Hey, does your config changed inputs work or do they also revert back after reopening the program? I opened a now closed issue report but the prob still remains.

have a look at  #14. Might be the cause for that. Does the default profile work?
Also are you on release 2.0+?

mKenfenheuer, 2022-06-10:
> The GUI looks great!
>
> Putting the version number in the log or somewhere else would help with bug reports I guess. I checked the version number of the executable (1.0) and under installed software [...]

Thats a good hint, will have a look at this, thanks!

mKenfenheuer, 2022-06-10:
Resolved in v0.2.3

https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/releases/tag/v0.2.3

closed 2022-06-10 by mKenfenheuer