mKenfenheuer/steam-deck-windows-usermode-driver #16: Trying to use the quick access button (3 dots btn) remap

AlexN85, 2022-06-10:
Hi I'm trying the make a function like the steam button and the game bar.

How do I make an action say to open the start menu or show desktop? 

Many thanks. I tried but failed :(

AlexN85, 2022-06-10:
Or only gamebar action has been implemented? And no other actions exist?

mKenfenheuer, 2022-06-10:
The actions work with the latest 0.1.x release, with 0.2.x its not yet implemented.

Dre2timez, 2022-06-10:
Hello, I have created a vid you can watch that describes how my inputs are not changing in the config file. 
https://youtu.be/CTO9veqpt2k

There's also this program I use very often for controllers called joyxoff which allows me to use the joystick as a mouse and the inputs collide with each other when using the steam decks controllers.

Gwizz007, 2022-06-12:
I can confirm this. 

What I did was open the config document located in documents as a txt file in notepad and manually changed and saved my changes there. To activate it I ran the program but when I closed it I clicked DONT save changes. So save note pad edit, don't save program changes. The program runs the note pad saved config. If you click save on the program it will revert to his default settings.

Use until fixed.

lwbt, 2022-06-13:
> Hello, I have created a vid you can watch that describes how my inputs are not changing in the config file.

> What I did was open the config document located in documents as a txt file in notepad and manually changed and saved my changes there.

@mKenfenheuer I think what they want to tell you is that the changes they try to make in the GUI are not saved to the configuration file. I noticed the same issue and wanted to file a new report but thought I'd better check open issues.

What I also noticed was that closing the GUI always asks if I want to save changes, even when I made no changes, but at the moment no changes are written and the file has to be edited by hand. One time I accidentally started the program twice, which is probably not a good for monitoring and writing changes to the configuration file but also handling the main driver functionality. But I may be too picky here, I leave it up to you. I can open a new issue if you want.

mKenfenheuer, 2022-06-13:
I've just published a new release which addresses the saving issue as well as the saving dialog.
Can you try the latest release please? If this still does not work, please create and upload a support package (available in the driver status page of new release)

lwbt, 2022-06-13:
I tried remapping buttons, which now works. What does not seem to work is saving changes to the configuration file or closing the GUI to let the driver run in the background. It continued to run in the background and asked for saving changes in 0.2.3. Now it seems to stop, the spinning Windows cursor briefly appears and the window is gone. I assume the program crashed, but I'm mostly a Linux user used to grepping logs and running programs from the command line to see if they really crashed in such situations, I may be wrong here. I can find event viewer logs from a few days ago when I was still using the portable single executable, but no recent logs which could give a hint if and why it crashes.

I'm on Windows 11.

It might be useful to log the Windows version the driver is running on and also provide a means to rotate or clear the debug log, which could grow significantly.

[SWICD_Support_637907283115345279.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/8892065/SWICD_Support_637907283115345279.zip)

mKenfenheuer, 2022-06-13:
@lwbt as i can see from your debug file, the configuration is corrupted.
According to the zip, the content is `SWICD_Lib.Config.Configuration`.
This might be the cause for the driver not exiting/saving your config file.

Can you delete the config file from your documents folder and try to save a new one?

lwbt, 2022-06-13:
> According to the zip, the content is `SWICD_Lib.Config.Configuration`.

It may sound silly, but that is not my configuration file.

> Can you delete the config file from your documents folder and try to save a new one?

I just did that and it crashes with the default configuration file when I want to close the GUI.

mKenfenheuer, 2022-06-15:
Then there can only be one reason. The configuration parser is located in a dll library, Maybe it was not updated for some reason.
Can you zip the program folder and upload it here?

To fix this, uninstall the whole driver (using control panel), delete any contents of the program folder if there is anything left and install it again.

lwbt, 2022-06-15:
> To fix this, uninstall the whole driver (using control panel), delete any contents of the program folder if there is anything left and install it again.

That worked. Now both, the default configuration and my custom configuration work. The GUI only prompts to write changes during closing when I made changes. No hangs, perfect! 

> Maybe it was not updated for some reason. Can you zip the program folder and upload it here?

Here you go: [steamdeck_issue16_configuration_parser.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/8914104/steamdeck_issue16_configuration_parser.zip)

mKenfenheuer, 2022-06-17:
Great, thank you. Indeed, the dll is from an older version and was not updated for some reason.
Glad this is fixed now.

closed 2022-06-17 by mKenfenheuer