mKenfenheuer/steam-deck-windows-usermode-driver #17: Lastest release buttons not inputting?

AlexN85, 2022-06-12:
Hi there

Love the new update. But one thing. Driver states it's started. The sound pops up. Windows and games acknowledges the contoller. But when I press anything nothing doesn't register? No buttons works. Any idea what I'm doing wrong. Thanks and always great work :)

mKenfenheuer, 2022-06-13:
@AlexN85 can you try the latest release please?
Maybe you need to change the operation mode (in settings) to whitelist or blacklist.

Can you confirm you can see something like `EmulationEnabled changed to: True` in the log?
[image]

If this is still not working, please upload the support package from the Driver status section (available in the new release).

AlexN85, 2022-06-14:
> @AlexN85 can you try the latest release please? Maybe you need to change the operation mode (in settings) to whitelist or blacklist.
>
> Can you confirm you can see something like `EmulationEnabled [...]

It worked!!! Changed to blacklist and it worked:)

By the way what does blacklist mean or whitelist mean? Why did I have to change that? :)

Awsome work my friend :)

mKenfenheuer, 2022-06-15:
There are three modes:

Blacklist: enables the driver if there is no blacklisted process running.
Whitelist: enables the driver if there is any whitelisted process running.
Combined: enables the driver if there is any whitelisted process and no blacklisted process running.

This allows you to only enable the driver if there is a certain game running and so on.

closed 2022-06-15 by mKenfenheuer