mKenfenheuer/steam-deck-windows-usermode-driver #21: [FEATURE] Keyboard actions and gui editor

mKenfenheuer, 2022-06-17:
**Is your feature request related to a problem? Please describe.**
Sometimes keyboard shortcuts are needed to close a game or to open a keyboard etc.

**Describe the solution you'd like**
Feature to have a button emulate a certain, user definable keyboard action like e.g. CTRL+ALT+DEL, ALT+F4 and so on.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

TankZealot, 2022-06-18:
Hey man! Firstly I have to say, thank you for this! Finally I can play some Windows games on the Steam Deck controller. 
However, I'm new to Github, does this mean that we're able to assign custom keyboard commands like ALT+F4 with the latest release or is this a feature that's coming?

mKenfenheuer, 2022-06-18:
Hi, thanks! No and yes. Its a feature i'm currently working on. 
My goal is basically to support keyboard only games and keyboard shortcuts in the future too.
e.g. Split/Second has really horrible controller support (only in game but not in menu, wtf.)
Or if your game does not have any exit feature, add an alt+f4 shortcut and bye bye :)

mKenfenheuer, 2022-06-26:
First implementations are now available in v0.2.5-beta1 which allows to map buttons to keyboard buttons.
Actions with multiple keys will follow

Gwizz007, 2022-07-03:
Just saw the long awaited keyboard update. Work has me tied up but I will give her a try soon. Thank you sooo much for your honest hard work!!!!

AshrafPatel, 2022-07-05:
How to map alt key I don't see it

Gwizz007, 2022-07-05:
On the note pad in documents, I noticed at the end he has instructions for Xbox shortcut overlay so I just followed that. Tried it with AMD adrenaline soverlay hortcuts but no success. Maybe it's still being fleshed out.

dot-bob, 2022-07-09:
Along with keyboard shortcuts perhaps a long press as an alternate function to some buttons like the "steam" and "..." buttons.  

The onexplayer uses long press to toggle alternate features.  For instance they have a keyboard button where a short press brings up the windows onscreen keyboard and a long press toggles mouse mode (something akin to lizard mode).  

A onscreen keyboard would sure make it handy for those games that require a keyboard to exit or bring up the taskbar.

Gwizz007, 2022-07-09:
Sounds like overworking the devs since we already have the full keyboard mappable and windows should be able to give you long press macros. At this point I think they should move to bug hunt mode or GUI simplifying.

AshrafPatel, 2022-07-18:
Keyboard shortcut to toggle lizard mode

Quazimojo, 2022-07-30:
I would be happy with just a button that would kick me out of a game to minimize it so i can adjust things like brightness etc.  

I havent found anything yet that would allow that

Either way thank you for helping making Windows on the deck a better experience.

closed 2022-09-19 by mKenfenheuer