mKenfenheuer/steam-deck-windows-usermode-driver #22: [BUG]Rear Grip Hardware button R5 does not seem to be registering input

CRISHTANG, 2022-06-19:
The Hardware Button R5  (lower right rear grip button) does not seem to be registering input when using SWICD.  All other hardware buttons seem to be functioning properly.  I tried assigning different functions to R5, checked app_config.conf for any anomalies, and started with freshly generated profiles but could not find a possible cause under the control of the user.  I verified that the R5 button on my Steam Deck unit activates properly under Steam; it is only SWICD where it does not function.

Reproduced by starting the SWICD program, making sure mode is set to 'blacklist';  Open Windows Game Controllers Control Panel (joy.cpl), to test buttons, only the R5 button does not register an input (default assignment given by SWICD is Windows Button 4/Y)

Driver version:  0.2.4-7f187ba-release

mKenfenheuer, 2022-06-20:
Hi,

please follow the issue template and upload a support package (available in driver status page). This includes your config, log file and the version of the driver.

CRISHTANG, 2022-06-22:
[SWICD_Support_637914714966529764.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/8954579/SWICD_Support_637914714966529764.zip)

mKenfenheuer, 2022-06-24:
Short update, ive been able to identify the bug.  I‘ll publish a fix with the next release asap. Thx for reporting!

mKenfenheuer, 2022-06-30:
Fixed in v0.2.5-beta1 upwards

closed 2022-06-30 by mKenfenheuer