mKenfenheuer/steam-deck-windows-usermode-driver #23: [BUG] B button acts as a start button?

AlexN85, 2022-06-20:
Hi I'm using the latest edition release and no matter what game I play the B button pauses the game and brings up the game menu? And therefore render that button in game can't be used.

Any way to not Have this happen. In the mapping screen it is as B button.

Thanks for all your hard work.

mKenfenheuer, 2022-06-20:
Hi,

please follow the issue template and upload a support package (available in driver status page). This includes your config, log file and the version of the driver.

My best guess right now is you should create a game specific profile and disable the lizard mode in the profile settings.
This is a typical behavior of the mouse and keyboard emulation (lizard mode) still enabled.

mKenfenheuer, 2022-06-30:
Closed due to inactivity.

closed 2022-06-30 by mKenfenheuer