mKenfenheuer/steam-deck-windows-usermode-driver #24: Could not open neptune controller

ReticentRobot, 2022-06-23:
**Describe the bug**
I just wiped my deck and reinstalled windows from scratch.  I've installed ViGEm, and SWICD but when I start it I'm getting an error and the controller is not recognized after the Driver is started.  The error states:

Could not open neptune controller: System.DllNotFoundException: Unable to load DLL 'hidapi.dll': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
   at hidapi.Native.HidApiNative.hid_enumerate(UInt16 vendor_id, UInt16 product_id)
   at hidapi.HidDevice.OpenDevice()
   at neptune_hidapi.net.NeptuneController.Open()
   at SWICD.Services.ControllerService.Start()

**To Reproduce**
Steps to reproduce the behavior:
1. Start Windows
2. Open SWICD Driver Control Panel
3. See error

**Expected behavior**
Driver loads and controller works

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Debug information**
Driver version: [v0.2.4-f187ba-release]

Create a support package by going to "Driver status" and selecting "Create support package".
Attach the file `SWICD_Support_XXXXX.zip` from your desktop.

[SWICD_Support_637916194649346935.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/8971745/SWICD_Support_637916194649346935.zip)

**Additional context**
Add any other context about the problem here.

ReticentRobot, 2022-06-24:
Troubleshooting attempted so far:

1) Uninstall ViGEm and SWICD Driver, reinstall both (no change)
2) Install Steam (no change)
3) Login and start Steam (no change)

ReticentRobot, 2022-06-24:
Still no luck.  I can confirm that the file was already there, I overwrote it with the one from the zip file anyway - but am still getting the error.  I also tried running the driver as Admin, but still same result.

ReticentRobot, 2022-06-24:
Not sure if this is even related or not (or helpful if it is), but the only thing I see in Event Viewer at the time of application launch is a warning as follows in the System log:

The application-specific permission settings do not grand Local Activation permission for the COM Server application with CLSID {2593F889-FEAF-457C-B68A-50F6B8EA6B54} and APPID {15C20B67-12E7-4BB6-92BB-7AFF07997402} to the user SteamDeck\ReticentRobot SID {...} from address LocalHost (Using LRPC) running in the application container Unavailable SID (Unavailable).  This security permission can be modified using the Component Services administrative tool.

mKenfenheuer, 2022-06-24:
I dont think this message is related. 
Did you try to run the release from the zip file?

ReticentRobot, 2022-06-24:
Yeah I unzipped and ran the exe from there, same thing.

mKenfenheuer, 2022-06-24:
Since the dll is built with MSVC, maybe the msvcr is missing? 

https://aka.ms/vs/17/release/vc_redist.x64.exe

https://docs.microsoft.com/de-de/cpp/windows/latest-supported-vc-redist?view=msvc-170#visual-studio-2015-2017-2019-and-2022

ReticentRobot, 2022-06-24:
That did the trick!  Thank you!! I'll go ahead and close as solved.  Very much appreciated for all of your work on this and the support.

closed 2022-06-24 by ReticentRobot