mKenfenheuer/steam-deck-windows-usermode-driver #27: [BUG] Latest version crashes randomly and shuts down

TankZealot, 2022-06-27:
**Describe the bug**
Hi!
I've been using 0.2.4 for a while and it has worked great! Tried the new beta and it also works great! However, the program sometimes crashes randomly. I tried reinstall everything and it still crashes. Went back to 0.2.4 and works without issue, but now I kinda miss the keyboard input.

**To Reproduce**
Steps to reproduce the behavior:
1. Just opening the program and then it will crash at random.

**Screenshots**
[image]

**Debug information**
New Beta

**Additional context**
Add any other context about the problem here.

TankZealot, 2022-06-27:
[SWICD_Support_637919294833284682.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/8991953/SWICD_Support_637919294833284682.zip)

DocMAX, 2022-07-10:
I can confirm that. Randomly the app closes and the joystick is gone.

mKenfenheuer, 2022-07-10:
Hi, thank you for reporting!
Looks like the driver is trying to write to the log file but ist somehow getting blocked by itself.

```
[27/06/2022 14:23:23][Critical]: System.IO.IOException: The process cannot access the file 'C:\Users\TommySteamDeck\Documents\SWICD\driver_log.log' because it is being used by another process.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
```

I suspect this is due to the debug output of the button states, can you confirm this happens if you are heavily pressing buttons or pressing some simultaneously? What if you start the driver and do not touch any buttons? Does it crash too?

mKenfenheuer, 2022-07-10:
Possibly fixed with release [v0.2.5-beta2](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/releases/tag/v0.2.5-beta2). Can you confirm?

DocMAX, 2022-07-10:
Looks stable now :-)

TankZealot, 2022-07-10:
Works much better for me too! Thank you for updating.

closed 2022-07-10 by mKenfenheuer