mKenfenheuer/steam-deck-windows-usermode-driver #28: [FEATURE] Delete Profile

ReticentRobot, 2022-07-08:
**Is your feature request related to a problem? Please describe.**
Cannot delete a profile once created

**Describe the solution you'd like**
Ability to delete a game profile

**Describe alternatives you've considered**

**Additional context**
Added SWICD.exe as a profile, would like to be able to delete that as it causes a bit of an issue :)

mKenfenheuer, 2022-07-10:
This is so obvious, how did i miss that ? 🤦‍♂️

mKenfenheuer, 2022-07-10:
Implemented with release [v0.2.5-beta2](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/releases/tag/v0.2.5-beta2)

closed 2022-07-10 by mKenfenheuer

ReticentRobot, 2022-07-11:
> This is so obvious, how did i miss that ? 🤦‍♂️

Lol, I didn't think about it either until I was dumb enough to add SWICD.exe to it 😂.  Thanks for getting this implemented so quickly!!