mKenfenheuer/steam-deck-windows-usermode-driver #29: [FEATURE]

ACCESSDENIIED, 2022-07-09:
**Is your feature request related to a problem? Please describe.**
Not a problem but definitely something i would love to be implemented.

**Describe the solution you'd like**
Curious whether you plan to add haptic feedback/rumble support?

**Describe alternatives you've considered**
GlosC/Glossi.

**Additional context**
While GlosC/Glossi have this feature, SWICD is so much more user friendly as I have it setup to whitelist playnite on startup and blacklist steam. This way I am able to boot into playnite to explore my entire library with addons that automatically change the refresh rate per game to what i've set them to. But I'm a big fan of rumble/haptic feedback so it's the one feature that's a personal deal breaker and I find myself torn between both this tool and GlosC.
As a final note, I want to thank you so much for this tool, It's the only one that is absolutely seamless and just works everytime with zero issues. It's also the only way for me to play World War Z with controller support as steam input and GlosC/Glossi do not work with that game unless i use M&K bindings.

mKenfenheuer, 2022-07-10:
Hi!,

indeed, rumble and haptic feedback is planned for future releases. I currently do not have much time due to work, but i'll see what i can do.

mKenfenheuer, 2022-09-21:
Implemented with v0.3.0

closed 2022-09-21 by mKenfenheuer