mKenfenheuer/steam-deck-windows-usermode-driver #30: [FEATURE] GPD style Keyboard-Mouse/Joystick toggle

DocMAX, 2022-07-10:
**Describe the solution you'd like**
A GPD style Keyboard-Mouse/Joystick toggle. I would like to be able to switch between Keyboard-Mouse and Jockstick with one defined button (the back buttons for example). Both should be available all the time, not depending on what process is running or not.

mKenfenheuer, 2022-07-10:
Good idea, maybe i can include this using the planned keyboard actions (e.g. pressing steam+x opens on screen keyboard, Steam + options toggles keyboard/mouse emulation and so on)

DocMAX, 2022-07-10:
Yeah that would be great. I my goal is to have 1 back button for switching kayboard/joystick mode and one for showing on screen keyboard EVERYWHERE. Still searching for a on 3rd party screen keyboard replacement with always on top/transparent features. Also individual hotkey actions for specific programs.

andyktaylor, 2022-07-26:
How about

https://joyxoff.com/en/

andyktaylor, 2022-07-28:
this works great in combination with your driver by the way. You can also enable/disable by holding down start+select, so not double-input issues

mKenfenheuer, 2022-09-21:
You can now toggle the lizard mode with the button actions

closed 2022-09-21 by mKenfenheuer