mKenfenheuer/steam-deck-windows-usermode-driver #31: [FEATURE] Minimize to tray on startup

ReticentRobot, 2022-07-11:
**Is your feature request related to a problem? Please describe.**
I would like to be able to have the driver application minimize to the tray icon rather than leave the window open on boot.

**Describe the solution you'd like**
Toggle in application (maybe next to the Start with Windows option to Minimize to Tray on Startup.  With this option selected the application will only appear as an icon in the task tray.  Clicking the task tray icon will bring up the Window as it does now when it is closed.

**Describe alternatives you've considered**
N/A

**Additional context**
Not a high priority, just a nice to have thing for aesthetic reasons (making Steam Deck more of a console like experience in Windows).

mKenfenheuer, 2022-07-11:
This should work from v0.2.5-beta2. There is an option to "Start Minimized"

[image]

Does this work for you?

ReticentRobot, 2022-07-11:
Oh wow, I haven't had a chance to install it yet - was planning on doing it on Wednesday and then updating the documentation with the changes.  It's like you're reading my mind before I even think it, lol!  I'll go ahead and close :)

closed 2022-07-11 by ReticentRobot