mKenfenheuer/steam-deck-windows-usermode-driver #34: [BUG] cant delete whitelist and blacklist items

andyktaylor, 2022-07-26:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

On default profile, click add whitelist, add a program.
Save profile

Now delete program from the default profile 
Save and exit
Log back in, program still shows 

Deleted and reinstalled software and the default profile remembers the program I've whitelisted.

Im on the latest beta version 

**Additional context**
Add any other context abo

chris-neely, 2022-08-05:
Same issue here, any work around d we can do while this is resolved? I really need to remove an item from whitelist.

mKenfenheuer, 2022-08-06:
As a workaround, you can edit the config file.
It's located in your documents folder > SWICD > app_config.conf.

Close the driver, open the file.
Then remove the corresponding whitelist=xxx line.
Save, restart the driver.

mKenfenheuer, 2022-08-06:
Please check the latest release, it should fix your issues.

https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/releases/tag/v0.2.7-beta2

andyktaylor, 2022-08-13:
Thanks, fixed

closed 2022-08-13 by mKenfenheuer