mKenfenheuer/steam-deck-windows-usermode-driver #35: [FEATURE] Add profile by typing executable name instead of using file picker

snailium, 2022-07-29:
**Is your feature request related to a problem? Please describe.**
PC Game Pass games has special protections in Windows 11. Even after acquiring ownership of WindowsApps folder, the file picker is still complaining about no file access permission when selecting game executable file.

**Describe the solution you'd like**
From the config file, it only stores executable file name. From the code, SWICD is looking through running processes (sorted A-Z?) and try to match executable file name. So, if we can create a profile by typing the file name instead of using the file picker, we should be able to create profiles for Game Pass games.

I have tried manually edit the config file and add ForzaHorizon5.exe. It seems SWICD is working as intended (disabling lizard mode).

**Describe alternatives you've considered**
Maybe choose file name by listing current running processes? So we can launch the game, and Alt-Tab to SWICD, list all running processes, and choose a process to create profile. In this way, we don't even need to research what the game executable file name is.

**Additional context**
N/A

mKenfenheuer, 2022-08-06:
I see, makes sense. I'll see what i can do.

mKenfenheuer, 2022-09-21:
Implemented

closed 2022-09-21 by mKenfenheuer