mKenfenheuer/steam-deck-windows-usermode-driver #36: [FEATURE] Lizard Mode Configuration

Mookers123, 2022-07-29:
**Is your feature request related to a problem? Please describe.**
When lizard mode is activated, the b and a buttons are jo longer useful in game, as they reproduce the escape/enter buttons.  If you turn off lizard mode to fix this problem, then you also lose the track pad mouse functionality.

**Describe the solution you'd like**
Please allow lizard mode to be configured such that I can keep the mouse functionality while adjusting and changing the other button functions.  Ie: a lizard mode where i can use just the track bad and right trigger for mouse functions, but all other buttons act like a joystick and dont emulate the keyboard.  Ie: allow us to configure lizard mode to adjust to our needs.

Alternatively, allow us to turn off lizard mode and then set up touchpads as mouse function.  Currently can only emulate 360 controller and doesn't emulate mouse unless lizard mode turned on.

**Additional context**
Thank you for a wonderful program!

mKenfenheuer, 2022-08-06:
Please check the latest release, it allows you to disable the lizard mode buttons and have custom mouse/keyboard mappings per profile.

https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/releases/tag/v0.2.7-beta2

delt31, 2022-08-08:
If we disable lizard mode for buttons, I lose control of all of my input and can’t control the deck anymore…

silveropensource, 2022-08-27:
You can't set the touchpad axis to be the mouse axis, so you can't reproduce the lizard mode with the current available options.

mKenfenheuer, 2022-09-21:
You can map the left and right mouse buttons to the trackpads in the default profile. Does that work for you?

mKenfenheuer, 2022-09-21:
This way you can have the touchpads act like a mouse like in the steamos desktop mode. Then you can use the other buttons for games. This way you can keep mouse functionality but also be able to game