mKenfenheuer/steam-deck-windows-usermode-driver #37: [FEATURE] Disable Lizard Mode through whitelist

rix1337, 2022-08-04:
My use case for this tool is running Forza Horizon 5 from the Windows store on my deck.
To make this as streamlined as possible I have set the tool to start with windows and whitelisted the ForzaHorizon5.exe
The detection and driver activation works as expected.
For Forza to actually pick up the driver though, lizard mode needs to be disabled through the UI.

This setting seems to apply globally right now, meaning Lizard mode is always off wenn running this tool.

I would love for the setting to also follow the whitelist, enabling me to use the Trackpad/Triggers for Mouse emulation when not running the game, while still being able to leave the tool running in the background.

mKenfenheuer, 2022-08-06:
For that, please create a profile for ForzaHorizon.exe, there you can disable the lizard mode for each executable.

rix1337, 2022-08-06:
What an obvious solution.
Thanks! Have just tested this, and it works ✅

closed 2022-08-06 by rix1337