mKenfenheuer/steam-deck-windows-usermode-driver #40: [BUG] v0.2.7-beta2 crashes with method not found exception

cherryduck, 2022-08-09:
**Describe the bug**
I was previously on v0.26-beta and updated to v0.2.7-beta2. My controls immediately stopped working. On opening up the SWICD control panel I can see there's a method not found exception for "boolean.neptunehidapi.net.NeptuneCo"...something, the rest of the message is off the screen. Unfortunately I can't investigate further because then the control panel goes completely unresponsive, and I can't even exit it when right clicking on the tray icon.

**To Reproduce**
Steps to reproduce the behavior:
1. Use SWICD
2. Crashes with error. Reboot doesn't fix it.

**Expected behavior**
The software not to crash with an error and lock up my controls.

**Debug information**
Driver version: v0.2.7-beta2 from here:
https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/releases/tag/v0.2.7-beta2

Unable to create a support package due to the control panel locking up upon crashing.

**Additional context**
Uninstalling this release and downgrading to the V0.26-beta resolves the issue.

mKenfenheuer, 2022-08-09:
Can you reinstall the 0.2.7 beta and see if its resolved? Seems like a dependent dll is not updated by the installer.

mKenfenheuer, 2022-08-09:
Also please upload the SWICD folder out of your documents as zip.
It should contain the log files as well as an app_config.conf file

cherryduck, 2022-08-09:
[SWICD.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/9294744/SWICD.zip)

Reinstalling 0.2.7 led to the same crash as before.

mKenfenheuer, 2022-08-13:
Did you uninstall first? Or just installed as update?

cherryduck, 2022-08-14:
Installed as an update.

nabel0, 2022-08-17:
[SWICD.ZIP](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/9365612/SWICD.ZIP)

nabel0, 2022-08-17:
Solved it by completely uninstalling and delting the folder in documents afterwards.
After an reinstallation the new version is then working fine

closed 2022-09-21 by mKenfenheuer