mKenfenheuer/steam-deck-windows-usermode-driver #41: [BUG] Yuzu emulator B acts as escape button

joemossjr16, 2022-08-09:
Yuzu emulator B acts as escape button. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to yuzu
2. Click configure
3. Hit b
4. The windows disappears
**Expected behavior**
To not disappear

joemossjr16, 2022-08-09:
Also when trying to add a profile a box pops up that asks me to type a name but doesnt do anything. and profiles arent saving

mKenfenheuer, 2022-08-09:
Please include a support package.

joemossjr16, 2022-08-09:
[SWICD_Support_637956534768805058.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/9294518/SWICD_Support_637956534768805058.zip)

joemossjr16, 2022-08-09:
Done

mKenfenheuer, 2022-08-09:
Can you try with v0.2.6 please? I have a bad feeling that the 0.2.7 .msi is broken

joemossjr16, 2022-08-09:
that works perfect now.

joemossjr16, 2022-08-09:
even the keyboard shortcuts in the back i set. Thanks man!

closed 2022-08-13 by mKenfenheuer

ToXIc-Dev, 2022-09-10:
Im still having this issue even when running v0.2.6