mKenfenheuer/steam-deck-windows-usermode-driver #45: Profile toggle via hotkey

wzzt, 2022-08-15:
Since the Deck doesn't use the ... button on Windows, I think it would be very convenient to use it to turn off/on SWICD inputs rather than add every Steam game someone owns to the blacklist or disable the driver altogether via the tray icon. Such a feature could also be used to toggle lizard mode or rearrange control setups on the fly, too.

mKenfenheuer, 2022-09-21:
Do the button actions in v0.3.0 provide what you need?

wzzt, 2022-09-22:
This is exactly what I needed. Thanks!

There does seem to be a small bug where deleting a button action doesn't actually remove it until I open the Edit dialog for another. I can easily reproduce this by setting a button to open Task Manager with LCtrl+LShift+Esc, then deleting the action.