mKenfenheuer/steam-deck-windows-usermode-driver #57: [FEATURE] Assign rumble (including intensity and period) to a button press in the Button Action GUI

andyktaylor, 2022-09-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Would be great to assign vibration/feedback to button presses

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

In the Button GUI, the ability to set a vibration when pressed.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

N/A - works well in Steam config, so would love to see it mirrored here

**Additional context**
Add any other context or screenshots about the feature request here.

mKenfenheuer, 2022-09-21:
I dont really get what your feature request is about. Do you want to have haptic feedback on a button press? Or do you want to toggle it via button presses?

andyktaylor, 2022-09-21:
Haptic vibration on a button press