mKenfenheuer/steam-deck-windows-usermode-driver #59: [BUG] left joystick go inversed sometimes

Phiinks, 2022-09-21:
**Describe the bug**
A clear and concise description of what the bug is.
in game on steamdeck (Win10), sometimes, the joystick go down when i push up

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
when i push up, the behavior in game is : going up

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Debug information**

latest version (0.3)

Create a support package by going to "Driver status" and selecting "Create support package".
Attach the file `SWICD_Support_XXXXX.zip` from your desktop.

**Additional context**
Add any other context about the problem here.

mKenfenheuer, 2022-09-21:
Please create a support package

joshuasjung, 2022-10-21:
Also seeing this issue.
Reproducible when making rapid movements with left joystick. Rolling motions from back side to forward

llmAsh, 2022-11-06:
Getting the same issue and I cannot fix it what's so ever. I've reinstalled windows and all drivers etc and it's still the same