mKenfenheuer/steam-deck-windows-usermode-driver #60: [BUG] Phantom Input In-Game Possibly Caused by Tetherscript HID Virtual Drivers

kevdunleavy, 2022-09-21:
**Describe the bug**
Some kind of repeated, unprompted button press, mouse movement, or scroll is happening in the menu of the Game Pass game _Danganronpa: Trigger Happy Havoc Anniversary Edition_. I don't believe this issue is game-specific though.

**To Reproduce**
Steps to reproduce the behavior:
1. Download and install _Danganronpa: Trigger Happy Havoc Anniversary Edition_ from _Game Pass for PC_
2. Start the game
3. Press A when prompted to get to the main menu
4. Observe the cursor flickering up and down constantly

**Expected behavior**
The cursor should stay on the first menu item while no inputs are performed.

**Screenshots**
Video: https://user-images.githubusercontent.com/1906657/191540316-14bca653-80c9-45b8-a911-d81952d6c728.mp4

**Debug information**
Driver version: `v0.3.0-0dc3c0b-release`

[SWICD_Support_637993689476265274.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/9617837/SWICD_Support_637993689476265274.zip)

**Additional context**

- When trying to figure out what was going on I uninstalled _Tetherscript HID Virtual Driver Kit_ and then retested the game. The issue was no longer happening and the game menu worked as expected.
- I also experienced this exact same issue with the menu of the Game Pass game _Yakuza: Like a Dragon_ but I did not investigate further before uninstalling the game so I don't know if uninstalling _Tetherscript HID Virtual Driver Kit_ would have solved the issue.
- I created [this Reddit thread](https://www.reddit.com/r/WindowsOnDeck/comments/xk2480/how_can_i_stop_this_phantom_scrollinginput_in/) looking for help with this issue before figuring out that it might be related to _Tetherscript HID Virtual Driver Kit_. Nobody else has commented there experiencing a similar yet issue but I'm linking the thread here just in case someone provides relevant info there in the future.

mKenfenheuer, 2022-09-21:
Workaround: In the programs folder of the tetherscript driver are batch files to uninstall certain drivers. Or you can use hidhide to hide the tetherscript joystick and gamepad. If you do not need keyboard / mouse emulation, you can leave the hid driver uninstalled. Its only needed for keyboard / mouse mappings

kevdunleavy, 2022-09-22:
Thanks @mKenfenheuer. I can confirm that disabling or uninstalling just the joystick driver is enough to resolve this issue. I ended up just disabling the joystick driver using the Control Panel. This way I get to keep the mouse mappings, which is great!

I am happy with this workaround but I'll leave it up to you to close this issue or keep it open since the underlying issue is unsolved. This is likely to catch other users out until the issue is either fixed or this workaround is documented.

Thanks for your help and the quick response 👍

Houfino, 2022-10-01:
Hi people...I have the same problem..I had to uninstall Tetherscript Virtual HID Driver Kit and solved...

[https://www.reddit.com/r/WindowsOnDeck/comments/xsas88/emulator_mesennes_setup_problem/](url)

Tetherscript Virtual HID is outdated version...Wants to patch newer versions :-)

GabrieleCurti, 2022-10-03:
I've also encountered the problem on some game pass games. Hollow knight would not accept any input, while Valheim was constantly getting the down input.

However uninstalling both the thetherscript gamepad driver and joystick driver fixed the issue, keeping keyboard emulation.
I am unsure which of the two is the driver that gives problem.

itsalexdc, 2022-10-25:
Hi,

I just want to help clearly guiding people through windows who are looking to fix this, as this was causing issues for me in Persona 5 and Tom Clancy's Wildlands on Windows 11.

You have to go to Settings >> System (on the left) >> About >> Device Manager >> Human Interface Devices >>
AND then delete Tetherscript Virtual Gamepad & Tetherscript Virtual Controller.