mKenfenheuer/steam-deck-windows-usermode-driver #63: [BUG]

dmj0shua1, 2022-09-29:
**Describe the bug**
When making a Button Action, 9 times out of 20, it won't ask if you wanna save the configuration when you close the window.

**To Reproduce**
Steps to reproduce the behavior:
1. Make a new button action
2. Close the app to minimize it to tray

**Expected behavior**
When closing the app, most times it would not ask you if you want to save your configuration.

**Debug information**
Driver version: v0.3.0-0dc3c0b-release

[SWICD_Support_638000716653373300.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/9678014/SWICD_Support_638000716653373300.zip)

dmj0shua1, 2022-09-29:
I meant to say 19 times out of 20.

Pherrera0, 2022-10-01:
I had the same issue. It happened after I changed the button assignments back to back a few times. Uninstalling, restarting, and then reinstalling SWICD and didn’t help. Deleting app_config.json in Documents>SWICD fixed it for me. I lost my config, but that’s not a big deal. 

This may also fix your other issue (#64).