mKenfenheuer/steam-deck-windows-usermode-driver #73: [BUG] Persona 5 Royal on Steam Deck WIN10 left pad issue

kosecki123, 2022-10-21:
**Describe the bug**
Game behaves like it left pad would always be enabled in up/left direction (the character in game always runs up left) 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to XBox app on Win10, run Persona 5 Royal

**Debug information**
Driver version: 0.3.0-0dc3c0b-release

Create a support package by going to "Driver status" and selecting "Create support package".
Attach the file `SWICD_Support_XXXXX.zip` from your desktop.

**Additional context**
Add any other context about the problem here.
[SWICD_Support_638019449972193654.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/9838213/SWICD_Support_638019449972193654.zip)

kosecki123, 2022-10-21:
Also tried latest GlosSI and it's the same, maybe issue is with the Vigem driver?

Phyore, 2022-10-24:
This is also occurring on Yakuza Like A Dragon (gamepass)

Phyore, 2022-10-24:
> Would the culprit for this. So it turns out that Tetherscript driver Joystick, Pad has been somehow enabled as player 2. Removing the device and driver from the system fixes the issue for me.

How did you go about doing this out of curiosity?

kosecki123, 2022-10-24:
> 

Actually I wanted to check what's causing the drift and went to https://gamepad-tester.com and then noticed that there is something under Player 2.

wefanqs, 2022-10-25:
> > Would the culprit for this. So it turns out that Tetherscript driver Joystick, Pad has been somehow enabled as player 2. Removing the device and driver from the system fixes the issue for me.
> 
> How did you go about doing this out of curiosity?

Appreciate your fix. I disabled device "Tetherscript Joystick" in Windows Device Manager and drifting issue gone.