mKenfenheuer/steam-deck-windows-usermode-driver #79: [BUG] A, B and Settings Buttons

FantaKlaaus, 2022-11-05:
**Describe the bug**
The A button seems to be locked onto ENTER as well as A on Xbox controller, so anytime you press A ingame it acts as both buttons. Same for the B and settings (triple line) buttons, but with ESCAPE instead of space. Can’t use B button ingame as it always brings up the main menu

**To Reproduce**
Steps to reproduce the behavior:
Use these buttons in a game

**Expected behavior**
The A button should only act as the A Xbox button, the B as the B button and the settings only as the settings button.

**Debug information**
Driver version: v0.3.2 -c049162

HD2MAX, 2022-11-07:
The problem seems to be related to the windows language, in Spanish there is this problem that button B presses ESC, and A - ENTER. in the emulators it is not usable, they go to window mode or the applications are closed.

Hopefully you can fix it soon. For now, the option is to put the **windows language in English** so that SWICD works correctly.

thanks for report first, I came to report this problem and I saw your report.

clemthi, 2022-11-08:
I don't think it's language related: I have a EN-US W10 installation and I had the same "issue".
[As explained in the wiki](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/wiki/Profiles#disable-lizard-mode): just disable the "Lizard mode" for the buttons in the profile you're using.