mKenfenheuer/steam-deck-windows-usermode-driver #81: [BUG] Tetherscript HID driver breaks Steam game controls even without SWICD running

HammerOfDaGods, 2022-11-07:
My use-case for SWICD is to only use it for non-Steam games and to use the Steam setup and drivers for Steam games. After installing the THID (Tetherscript HID drivers), even with SWICD closed and not running my controls are completely non-functional in Steam games (The example here is Mass Effect 3/Mass Effect Legendary Edition). After some troubleshooting I noticed that with THID installed, Steam detects it as two additional controllers (Tetherscript Virtual Gamepad and Tetherscript Virtual Joystick). After trying the game with THID installed as well as uninstalled it seems to work fine as long as it's not installed. Is there an easy way to disable/enable the THID dirvers or hide them from Steam? As it stands now the only thing I can think to do install THID when I need it and uninstall it when I don't, but that's far from a good solution.

HammerOfDaGods, 2022-11-14:
Quite disappointed that this got ignored, especially considering how little work it would have taken. I've already found a workaround that could easily be integrated into your program. You literally just need a button that runs the scripts that are included in the Tetherscript drivers. In the install folder there is an "InstallAll.bat" and "UninstallAll.bat" that will quickly and cleanly remove or add the drivers as needed. Admin privilege required, but you can just have the button run as admin and the user would click yes on the prompt.

GoldenReverie, 2022-12-06:
can confirm this driver also breaks game pass games.