mKenfenheuer/steam-deck-windows-usermode-driver #82: [BUG] Vibration on Right Stick touch

inCrooo, 2022-11-08:
**Describe the bug**
-There's a light vibration when I touch the right stick. 
-It doesn't happen on the left stick, only the right.
-"Right stick touch" is not mapped to anything.
-The vibration on right stick touch persists even after disabling haptic feedback.

**Expected behavior**
No vibration on stick touch.

**Debug information**
Driver version: v0.3.2

**Additional context**
Add any other context about the problem here.
[SWICD_Support_638035180799042692.zip](https://github.com/mKenfenheuer/steam-deck-windows-usermode-driver/files/9962677/SWICD_Support_638035180799042692.zip)

redphx, 2022-11-09:
I also have this problem on 0.3.2 after installing tetherscript. It happens on both sticks for me though. I tried to use the default `app_config.json` file but it doesn't fix the issue.

inCrooo, 2022-11-09:
> I also have this problem on 0.3.2 after installing tetherscript. It happens on both sticks for me though. I tried to use the default `app_config.json` file but it doesn't fix the issue. I'm using Win 11 22H2.

You're right. It's both sticks. But for some reason the right stick vibration is much more noticeable.

I do have tetherscript installed. I uninstalled all tetherscript drivers one by one but the vibration still persists. I also tried default 
app_config.json and it didn't help.

The only thing that stops the vibration is pausing/closing SWICD.

I'm on Windows 10 21H2.

inCrooo, 2022-11-09:
Sorry didn't mean to close it.

redphx, 2022-11-13:
Did a clean Win 11 install and still have this bug.