mKenfenheuer/steam-deck-windows-usermode-driver #90: tetherscript virtual hid driver kit not found

MrKira34, 2022-12-10:
https://tetherscript.com/hid-driver-kit-download/
Page not fount

T-Niijima, 2022-12-17:
https://tetherscript.com/hid-blog/hvdk-now-discontinued/

HVDK is discontinued. 
We have to find a way to substitute

wemod-deck, 2022-12-30:
Mirror: https://web.archive.org/web/20220926192425/https://tetherscript.s3-us-west-2.amazonaws.com/HVDK/HVDK+Standard_2.1_Installer.exe

LukeDevsE, 2023-01-02:
the one wemod-deck stops randomly.
heres a link that actually works 
[https://www.mediafire.com/file/47p107w44tsjfyf/tether.exe/file](url)

wemod-deck, 2023-01-03:
> the one @wemod-deck gave stops downloading at 15.7 mbs. heres a link that actually works https://www.mediafire.com/file/47p107w44tsjfyf/tether.exe/file

Could you also upload it to archive.org?

LukeDevsE, 2023-01-07:
https://archive.org/details/tether_202301

niratama, 2023-01-15:
https://github.com/tetherscript/hvdk
> You can still obtain the Tetherscript-signed drivers by downloading ControlMyJoystick 14-day free trial. The drivers will continue to work even if you do not purchase ControlMyJoystick. There is no longer a standalone drivers download available.